xreef/EByte_LoRa_E220_Series_Library #1: Receive with RSSI bug

xulioc, 2022-04-14:
https://github.com/xreef/EByte_LoRa_E220_Series_Library/blob/4e3a13e87cf0e098ff831974f17c094f37b7bee8/LoRa_E220.cpp#L745

Arduino substring is exclusive for the second argument, so should be:
 rc.data = tmpData.substring(0, tmpData.length());

xreef, 2022-04-16:
Hi @xulioc,
yes, we [noticed that problem some days ago](https://www.mischianti.org/forums/topic/bug-when-sending-number-with-lora_e220-sendmessage/), but now I'm not at home, so I think I retest all in the following weeks.

https://www.mischianti.org/forums/topic/bug-when-sending-number-with-lora_e220-sendmessage/

Thanks for the alert, bye Renzo

xreef, 2022-04-19:
Hi @xulioc,
I can't replicate the problem.

can you explain better what happened for you and with Arduino you use.

I tested RSSI with Arduino Nano 33 IoT and an STM32F103, and all worked correctly.

Bye Renzo

closed 2022-06-06 by xreef