xreef/EByte_LoRa_E220_Series_Library #2: setConfiguration and "check wiring"

ChadMatsalla, 2022-06-28:
I'm using a E220-900T22D.
Version: E220-900T22D-V1.X

I'm using an esp32.

I'm trying to use your example SetConfiguration sketch.

If the program looks like this:
```
#include "Arduino.h"
#include "LoRa_E220.h"

HardwareSerial serial2(1);
LoRa_E220 e220ttl(32, 33, &serial2, 35, 26, 25, UART_BPS_RATE_9600, SERIAL_8N1);

void printParameters(struct Configuration configuration);
void printModuleInformation(struct ModuleInformation moduleInformation);

void setup() {
    Serial.begin(115200);
    e220ttl.begin();
}
void loop() {
}
```
I see:
```
RX MIC ---> 32
TX MIC ---> 33
AUX ---> 35
M0 ---> 26
M1 ---> 25
Init AUX pin!
Init M0 pin!
Init M1 pin!
Begin ex
Begin Hardware Serial
PIN SELECTED!!
Begin 
MODE NORMAL!
AUX HIGH!
Complete!
```

This seems good. But If I run the entire program using the constructor above and using this as loop():
```
void setup() {
	Serial.begin(115200);
	e220ttl.begin();
	ResponseStructContainer c;
	c = e220ttl.getConfiguration();
	Configuration configuration = *(Configuration*) c.data;
	Serial.println(c.status.getResponseDescription());
	Serial.println(c.status.code);

	printParameters(configuration);

	ResponseStatus rs = e220ttl.setConfiguration(configuration, WRITE_CFG_PWR_DWN_SAVE);
	Serial.println(rs.getResponseDescription());
	Serial.println(rs.code);

	c = e220ttl.getConfiguration();
	configuration = *(Configuration*) c.data;
	Serial.println(c.status.getResponseDescription());
	Serial.println(c.status.code);

	printParameters(configuration);
	c.close();
  */
}
```
I see a ton of things as follows:
```
HEAD : EB DE C3
 
AddH : BF
AddL : 6E
 
Chan : 132 -> 542MHz
 
SpeedParityBit     : 10 -> 8E1
SpeedUARTDatte     : 101 -> 38400bps
SpeedAirDataRate   : 101 -> 19.2kbps
 
OptionSubPacketSett: 10 -> 64bytes
OptionTranPower    : 1 -> 17dBm
OptionRSSIAmbientNo: 1 -> Enabled
 
TransModeWORPeriod : 111 -> 4000ms
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeFixedTrans: 1 -> Fixed transmission (first three bytes can be used as high/low address and channel)
----------------------------------------
MODE NORMAL!
AUX HIGH!
Complete!
No response from device! (Check wiring)
12
----------------------------------------
HEAD : EB DE C3
 
AddH : BF
AddL : 6E
 
Chan : 132 -> 542MHz
 
SpeedParityBit     : 10 -> 8E1
SpeedUARTDatte     : 101 -> 38400bps
SpeedAirDataRate   : 101 -> 19.2kbps
 
OptionSubPacketSett: 10 -> 64bytes
OptionTranPower    : 1 -> 17dBm
OptionRSSIAmbientNo: 1 -> Enabled
 
TransModeWORPeriod : 111 -> 4000ms
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeFixedTrans: 1 -> Fixed transmission (first three bytes can be used as high/low address and channel)
----------------------------------------
MODE SLEEP CONFIG!
AUX HIGH!
Complete!
AUX HIGH!
Complete!
Clear buffer...ok!
Available buffer: 0 structure size: 11
----------------------------------------
HEAD : C0 0 8
 
AddH : 0
AddL : 3
 
Chan : 23 -> 433MHz
 
SpeedParityBit     : 0 -> 8N1 (Default)
SpeedUARTDatte     : 11 -> 9600bps (default)
SpeedAirDataRate   : 10 -> 2.4kbps (default)
 
OptionSubPacketSett: 0 -> 200bytes (default)
OptionTranPower    : 0 -> 22dBm (Default)
OptionRSSIAmbientNo: 0 -> Disabled (default)
 
TransModeWORPeriod : 11 -> 2000ms (default)
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeFixedTrans: 0 -> Transparent transmission (default)
----------------------------------------
MODE NORMAL!
AUX HIGH!
Complete!
Save mode returned not recognized!
11
MODE SLEEP CONFIG!
AUX HIGH!
Complete!
3
Available buffer: 0 structure size: 11
----------------------------------------
HEAD : 53 61 76
 
AddH : 65
AddL : 20
 
Chan : 100 -> 510MHz
 
SpeedParityBit     : 1 -> 8O1
SpeedUARTDatte     : 11 -> 9600bps (default)
SpeedAirDataRate   : 101 -> 19.2kbps
 
OptionSubPacketSett: 1 -> 128bytes
OptionTranPower    : 11 -> 10dBm
OptionRSSIAmbientNo: 1 -> Enabled
 
TransModeWORPeriod : 101 -> 3000ms
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeFixedTrans: 1 -> Fixed transmission (first three bytes can be used as high/low address and channel)
----------------------------------------
MODE NORMAL!
AUX HIGH!
Complete!
No response from device! (Check wiring)
12
----------------------------------------
HEAD : 53 61 76
 
AddH : 65
AddL : 20
 
Chan : 100 -> 510MHz
 
SpeedParityBit     : 1 -> 8O1
SpeedUARTDatte     : 11 -> 9600bps (default)
SpeedAirDataRate   : 101 -> 19.2kbps
 
OptionSubPacketSett: 1 -> 128bytes
OptionTranPower    : 11 -> 10dBm
OptionRSSIAmbientNo: 1 -> Enabled
 
TransModeWORPeriod : 101 -> 3000ms
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeFixedTrans: 1 -> Fixed transmission (first three bytes can be used as high/low address and channel)
----------------------------------------
```
I've checked the wiring and used two different E220s and two different esp32s.

Can you please help me understand what I'm doing wrong?

Thanks,

Chad Matsalla

xreef, 2022-06-28:
Hi Chad,
probably It's a power supply problem, try to add 5v and good amperage to the device and give me feedback.
Bye Renzo

ChadMatsalla, 2022-06-28:
This is unrelated, but I can't get the two systems to communicate. They have the same configurations:
Client:
```
AddH : 0
AddL : 0
 
Chan : 23 -> 433MHz
 
SpeedParityBit     : 0 -> 8N1 (Default)
SpeedUARTDatte     : 11 -> 9600bps (default)
SpeedAirDataRate   : 10 -> 2.4kbps (default)
 
OptionSubPacketSett: 0 -> 200bytes (default)
OptionTranPower    : 0 -> 22dBm (Default)
OptionRSSIAmbientNo: 0 -> Disabled (default)
 
TransModeWORPeriod : 11 -> 2000ms (default)
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeFixedTrans: 0 -> Transparent transmission (default)
```
Gateway:
```
AddH : 0
AddL : 0
 
Chan : 23 -> 433MHz
 
SpeedParityBit     : 0 -> 8N1 (Default)
SpeedUARTDatte     : 11 -> 9600bps (default)
SpeedAirDataRate   : 10 -> 2.4kbps (default)
 
OptionSubPacketSett: 0 -> 200bytes (default)
OptionTranPower    : 0 -> 22dBm (Default)
OptionRSSIAmbientNo: 0 -> Disabled (default)
 
TransModeWORPeriod : 11 -> 2000ms (default)
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeFixedTrans: 0 -> Transparent transmission (default)
```
I don't have actual antennae - I inserted a 3.25" piece of wire into the SMC connector. They are sitting ajdacent to each other in a breadboard.

I found no information in whether you can use these without an antenna.

Do you have any suggestions?

Thank you!

closed 2022-07-21 by xreef