xreef/EByte_LoRa_E220_Series_Library #6: EByte E220 only receiving blank characters

drank40, 2022-10-06:
I have 2 E220's set up with an esp32 and an esp8266(-01) respectively.

I connected the modules as follows:

<figure class="wp-block-table mischianti-table-left"><table><thead><tr><th>E220</th><th>esp32</th></tr></thead><tbody><tr><td>M0</td><td>21</td></tr><tr><td>M1</td><td>19</td></tr><tr><td>TX</td><td>16 (PullUP 4,7KΩ)</td></tr><tr><td>RX</td><td>17 (PullUP 4,7KΩ)</td></tr><tr><td>AUX</td><td>PIN 15 (PullUP 4,7KΩ)</td></tr><tr><td>VCC</td><td> 3.3v</td></tr><tr><td>GND</td><td>GND</td></tr><tr><td></td><td></td></tr></tbody></table></figure>

<figure class="wp-block-table mischianti-table-left"><table><thead><tr><th>E220</th><th>esp8266</th></tr></thead><tbody><tr><td>M0</td><td>GND</td></tr><tr><td>M1</td><td>GND</td></tr><tr><td>TX</td><td>GPIO 1 (PullUP 4,7KΩ)</td></tr><tr><td>RX</td><td>GPIO 3 (PullUP 4,7KΩ)</td></tr><tr><td>AUX</td><td>GPIO 2 (PullUP 4,7KΩ)</td></tr><tr><td>VCC</td><td>3.3v</td></tr><tr><td>GND</td><td>GND</td></tr><tr><td></td><td></td></tr></tbody></table></figure>

And the esps are running these programs : 
ESP32 (receiver):

```cpp
#include <Arduino.h>;
#include <LoRa_E220.h>;

// 16 rx 17 tx

LoRa_E220 e220ttl(&Serial2, 15, 21, 19, UART_BPS_RATE_115200); //  RX AUX M0 M1
 
void setup() {
  Serial.begin(115200);
  Serial.println();
  pinMode(LED_BUILTIN, OUTPUT);
  delay(500);
 
  Serial.println(F("Node online. Attempting to ping..."));
 
  // Startup all pins and UART
  e220ttl.begin();
 
  // Send message
  ResponseStatus rs = e220ttl.sendMessage(F("Hello, world?"));
  // Check If there is some problem of successfully send
  Serial.println(rs.getResponseDescription());
    digitalWrite(LED_BUILTIN, HIGH);
}
 
void loop() {
    // If something available
  if (e220ttl.available()>1) {
      // read the String message
    ResponseContainer rc = e220ttl.receiveMessage();
    // Is something goes wrong print error
    if (rc.status.code!=1){
        rc.status.getResponseDescription();
    }else{
        // Print the data received
        Serial.println(rc.data);
        Serial.println(rc.data.length());
        digitalWrite(LED_BUILTIN, LOW);
    }
  }
  if (Serial.available()) {
        String input = Serial.readString();
        e220ttl.sendMessage(input);
        Serial.println("Sent.");
  }
}

```

ESP8266 (transmitter):

```cpp
#include <Arduino.h>;
#include <LoRa_E220.h>;

// 16 rx 17 tx

LoRa_E220 e220ttl(&Serial, 2, 21, 19, UART_BPS_RATE_115200); //  RX AUX M0 M1
 
void setup() {
  Serial.begin(115200);
  Serial.println();
  pinMode(LED_BUILTIN, OUTPUT);
  delay(500);
 
  Serial.println(F("Node online. Attempting to ping..."));
 
  // Startup all pins and UART
  e220ttl.begin();
 
  // Send message
  ResponseStatus rs = e220ttl.sendMessage(F("Hello, world?"));
  // Check If there is some problem of successfully send
  Serial.println(rs.getResponseDescription());
    digitalWrite(LED_BUILTIN, HIGH);
}
 
void loop() {
    e220ttl.sendMessage(F("Hello world!"));
    delay(2000);
}
```

From this code I would expect the esp32 to print to the serial console the message transmitted by the esp8266 along with its size every 2 seconds, but what I get is this:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4

Node online. Attempting to ping...
Success

564

561

569

578

572

577

565
```

and so on. I just get a very long string made up of blank chracters as output, and this only happens when the transmitter is plugged in, so it isnt noise.

I also tried powering the modules with 5v from the esp32, with the same result. Ill attach some pictures of my setup in case the wiring is wrong.

Ive looked through many forum posts but I havent read anything similar to what im experiencing, I'm hoping you can help.

Thanks for your hard work on the library.

[image]

[image]

PS: the pullup resistors are soldered onto the perfboard

xreef, 2022-10-06:
Hi,
you can only send String and char*, you can't use __FlashStringHelper .

Bye Renzo

drank40, 2022-10-07:
Removing the F() did not change the output. Any ideas?

xreef, 2022-10-12:
Hi @drank40,
try to use the standard baud rate, remove UART_BPS_RATE_115200 and give me feedback.
Bye RM

closed 2023-01-13 by xreef

Tatbiq-Tech, 2024-07-27:
Hi, what if my board supports only UART_BPS_RATE_115200  ??
default rate does not work at me end

Tatbiq-Tech, 2024-07-27:
@drank40 I know it has been 2 years, but how did you solve your issue ?
I'm facing the same exact issue, receiver shows blank data.

drank40, 2024-08-12:
> @drank40 I know it has been 2 years, but how did you solve your issue ? I'm facing the same exact issue, receiver shows blank data.

Hi, I dont have access to the project now but I think it was an hw issue, because I rebuilt it and uploaded a similar code and it works now