xreef/EByte_LoRa_E220_Series_Library #8: The #define FREQUENCY_XXX parameter does not change the displayed value of the F

WRoss7, 2022-12-09:
I am not an expert, but I found that by editing the third line of the "statesNaming.h" file as follows, the frequency is correctly displayed:

#if defined(FREQUENCY_433)

xreef, 2022-12-14:
Hi WRoss7,
Sorry, I lost this issue, you can follow this topic to change frequencies.
https://www.mischianti.org/forums/topic/esp8266-wemos-d1-mini-and-e220-900t30d-frequencies-problem-and-info/
Bye Renzo

closed 2022-12-14 by xreef

WRoss7, 2022-12-14:
Ciao Renzo,
Perhaps I did not explain myself well: I had inserted the line #define FREQUENCY_868 from the very beginning on my code, but the frequency displayed was always 433 MHz. Then I checked in the library files and found that by editing the third line in the "/includes/statesNaming.h" file as shown below, and recompiled the code, the displayed frequency appears correct: 868 MHz.

Before
#ifdef FREQUENCY_433

After
#if defined(FREQUENCY_433)

I understood it as an oversight, since the correction I made worked.

P.S. really nice gadget, thanks for your work