xreef/EByte_LoRa_E220_Series_Library #9: Is the variable to set the encryption/decryption key missing?

WRoss7, 2022-12-11:
In the library I see the handling of encryption/decryption, maybe I am doing something wrong? I am not an expert in this area.

xreef, 2022-12-12:
Hi WRoss7,
the enc/dec work without any functions. You can change the key by changing the [CRYPT key](https://github.com/xreef/EByte_LoRa_E220_Series_Library/blob/master/LoRa_E220.h#L200).
Bye Renzo

closed 2022-12-12 by xreef

WRoss7, 2022-12-12:
Thank you for the quick reply.

Do you intend to enter a value for these two fields directly into the library?
struct Crypt {
	byte CRYPT_H = 0;
	byte CRYPT_L = 0;
};

Is there any chance that a variable (as for #define FREQUENCY_xxx) will be made available in a future version?

xreef, 2022-12-12:
Hi,
no, the define interacts with my code by enabling some descriptions; Crypt is a module parameter for hardware encryption.
Bye Renzo