xreef/EByte_LoRa_E220_Series_Library #12: waitCompleteResponse

aschewski, 2023-02-04:
Hello Renzo. I had an issue where if I sent multiple messages within a few seconds, they would not arrive. I then set line 436 in LoRa_E220.cpp to result = this->waitCompleteResponse(200, 200). Now I can send messages quickly one after another. However, I don't understand the wait time? What is the purpose of the first and second value? Could you please explain it in detail? I would like to understand what negative effects reducing the values can have. Thank you in advance and greetings.

xreef, 2023-02-04:
Hi aschewski,
the first is needed if AUX never goes HIGH, and the second is used if you don't connect the AUX.

But if you don't connect the AUX not change the second value, if works for you a smaller value, It's only lucky.

If you need a faster response connect AUX, there isn't another way.

Bye Renzo

closed 2023-02-07 by xreef

aschewski, 2023-02-26:
OK! Thank you - I have now connected AUX and it works great. Thanks for the great library.