xreef/EByte_LoRa_E220_Series_Library #13: How to configuration with ESP32-S3 Board?

blacxcode, 2023-05-02:


xreef, 2023-05-02:
Hi Blacxcode,
you can use the same wiring of esp32, more information on https://www.mischianti.org/category/my-libraries/lora-e32-devices/
Bye Renzo

closed 2023-05-02 by xreef

blacxcode, 2023-05-05:
Hi, xreef

I think esp32-s3 pinout is different from esp32.
[https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/_images/ESP32-S3_DevKitC-1_pinlayout_v1.1.jpg](url)

I can't find TX2 and RX2 pin information in esp32-s3, so I can't use serial2. Do you know any alternative solution for this case?
Translated with DeepL https://www.deepl.com/app/?utm_source=android&utm_medium=app&utm_campaign=share-translation

blacxcode, 2023-05-05:
Hi, xreef

I think esp32-s3 pinout is different from esp32.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/_images/ESP32-S3_DevKitC-1_pinlayout_v1.1.jpg
I can't find TX2 and RX2 pin information in esp32-s3, so I can't use serial2. Do you know any alternative solution for this case?