xreef/EByte_LoRa_E220_Series_Library #14: In the example program getConfiguration getting error "Data size not match" on esp32-s3 board

blacxcode, 2023-05-06:
Hi Xreef, 
I've tried the getConfiguration code sample with the ESP32-S3 board, but I'm getting an error "Data size not match! 7". Then I compared it to trying on an ESP32 board with the same code sample and it worked get return "Succees 1". 

I am using UART2 ESP32-S3 configuration based on HardwareSerial.h
https://github.com/espressif/arduino-esp32/edit/master/cores/esp32/HardwareSerial.cpp

```
#if SOC_UART_NUM > 2
#ifndef RX2
#if CONFIG_IDF_TARGET_ESP32
#define RX2 16
#elif CONFIG_IDF_TARGET_ESP32S3
#define RX2 19 
#endif
#endif

#ifndef TX2
#if CONFIG_IDF_TARGET_ESP32
#define TX2 17
#elif CONFIG_IDF_TARGET_ESP32S3
#define TX2 20
#endif
#endif
```

This my code :

```
/*
   LoRa E220
   Get configuration.
   You must uncommend the correct constructor.

   by Renzo Mischianti <https://www.mischianti.org>

   https://www.mischianti.org
*/

#include "Arduino.h"
#include "LoRa_E220.h"

//ESP32
#define PIN_ESP32_RX2    16
#define PIN_ESP32_TX2    17
#define PIN_ESP32_AUX   18
#define PIN_ESP32_M0    21
#define PIN_ESP32_M1    19

//ESP32-S3
#define PIN_ESP32S3_RX2    19
#define PIN_ESP32S3_TX2    20
#define PIN_ESP32S3_AUX   11
#define PIN_ESP32S3_M0    12
#define PIN_ESP32S3_M1    13

// ---------- ESP32 pins ---------------
//LoRa_E220 e220ttl(PIN_ESP32_RX2, PIN_ESP32_TX2, &Serial2, PIN_ESP32_AUX, PIN_ESP32_M0, PIN_ESP32_M1, UART_BPS_RATE_9600, SERIAL_8N1); //  esp32
// -------------------------------------

// ---------- ESP32-S3 pins ------------
LoRa_E220 e220ttl(PIN_ESP32S3_RX2, PIN_ESP32S3_TX2, &Serial2, PIN_ESP32S3_AUX, PIN_ESP32S3_M0, PIN_ESP32S3_M1, UART_BPS_RATE_9600, SERIAL_8N1); //  esp32-s3 
// -------------------------------------

void printParameters(struct Configuration configuration);
void printModuleInformation(struct ModuleInformation moduleInformation);

void setup() {
  Serial.begin(115200);
  //Serial2.begin(9600, SERIAL_8N1, PIN_ESP32_RX2, PIN_ESP32_TX2); //ESP32
  Serial2.begin(9600, SERIAL_8N1, PIN_ESP32S3_RX2, PIN_ESP32S3_TX2); //ESP32-S3

  while (!Serial) {};
  delay(500);

  Serial.println();

  // Startup all pins and UART
  e220ttl.begin();

  ResponseStructContainer c;
  c = e220ttl.getConfiguration();
  // It's important get configuration pointer before all other operation
  Configuration configuration = *(Configuration*) c.data;
  Serial.println(c.status.getResponseDescription());
  Serial.println(c.status.code);

  printParameters(configuration);

  ResponseStructContainer cMi;
  cMi = e220ttl.getModuleInformation();
  // It's important get information pointer before all other operation
  ModuleInformation mi = *(ModuleInformation*)cMi.data;

  Serial.println(cMi.status.getResponseDescription());
  Serial.println(cMi.status.code);

  printModuleInformation(mi);

  c.close();
  cMi.close();
}

void loop() {

}

void printParameters(struct Configuration configuration) {
  Serial.println("----------------------------------------");

  Serial.print(F("HEAD : "));  Serial.print(configuration.COMMAND, HEX); Serial.print(" "); Serial.print(configuration.STARTING_ADDRESS, HEX); Serial.print(" "); Serial.println(configuration.LENGHT, HEX);
  Serial.println(F(" "));
  Serial.print(F("AddH : "));  Serial.println(configuration.ADDH, HEX);
  Serial.print(F("AddL : "));  Serial.println(configuration.ADDL, HEX);
  Serial.println(F(" "));
  Serial.print(F("Chan : "));  Serial.print(configuration.CHAN, DEC); Serial.print(" -> "); Serial.println(configuration.getChannelDescription());
  Serial.println(F(" "));
  Serial.print(F("SpeedParityBit     : "));  Serial.print(configuration.SPED.uartParity, BIN); Serial.print(" -> "); Serial.println(configuration.SPED.getUARTParityDescription());
  Serial.print(F("SpeedUARTDatte     : "));  Serial.print(configuration.SPED.uartBaudRate, BIN); Serial.print(" -> "); Serial.println(configuration.SPED.getUARTBaudRateDescription());
  Serial.print(F("SpeedAirDataRate   : "));  Serial.print(configuration.SPED.airDataRate, BIN); Serial.print(" -> "); Serial.println(configuration.SPED.getAirDataRateDescription());
  Serial.println(F(" "));
  Serial.print(F("OptionSubPacketSett: "));  Serial.print(configuration.OPTION.subPacketSetting, BIN); Serial.print(" -> "); Serial.println(configuration.OPTION.getSubPacketSetting());
  Serial.print(F("OptionTranPower    : "));  Serial.print(configuration.OPTION.transmissionPower, BIN); Serial.print(" -> "); Serial.println(configuration.OPTION.getTransmissionPowerDescription());
  Serial.print(F("OptionRSSIAmbientNo: "));  Serial.print(configuration.OPTION.RSSIAmbientNoise, BIN); Serial.print(" -> "); Serial.println(configuration.OPTION.getRSSIAmbientNoiseEnable());
  Serial.println(F(" "));
  Serial.print(F("TransModeWORPeriod : "));  Serial.print(configuration.TRANSMISSION_MODE.WORPeriod, BIN); Serial.print(" -> "); Serial.println(configuration.TRANSMISSION_MODE.getWORPeriodByParamsDescription());
  Serial.print(F("TransModeEnableLBT : "));  Serial.print(configuration.TRANSMISSION_MODE.enableLBT, BIN); Serial.print(" -> "); Serial.println(configuration.TRANSMISSION_MODE.getLBTEnableByteDescription());
  Serial.print(F("TransModeEnableRSSI: "));  Serial.print(configuration.TRANSMISSION_MODE.enableRSSI, BIN); Serial.print(" -> "); Serial.println(configuration.TRANSMISSION_MODE.getRSSIEnableByteDescription());
  Serial.print(F("TransModeFixedTrans: "));  Serial.print(configuration.TRANSMISSION_MODE.fixedTransmission, BIN); Serial.print(" -> "); Serial.println(configuration.TRANSMISSION_MODE.getFixedTransmissionDescription());

  Serial.println("----------------------------------------");
}

void printModuleInformation(struct ModuleInformation moduleInformation) {
  Serial.println("----------------------------------------");
  Serial.print(F("HEAD: "));  Serial.print(moduleInformation.COMMAND, HEX); Serial.print(" "); Serial.print(moduleInformation.STARTING_ADDRESS, HEX); Serial.print(" "); Serial.println(moduleInformation.LENGHT, DEC);

  Serial.print(F("Model no.: "));  Serial.println(moduleInformation.model, HEX);
  Serial.print(F("Version  : "));  Serial.println(moduleInformation.version, HEX);
  Serial.print(F("Features : "));  Serial.println(moduleInformation.features, HEX);
  Serial.println("----------------------------------------");
}
```

Serial Monitor :

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4

RX MIC ---> 19
TX MIC ---> 20
AUX ---> 11
M0 ---> 12
M1 ---> 13
Init AUX pin!
Init M0 pin!
Init M1 pin!
Begin ex
Begin Hardware Serial
PIN SELECTED!!
Begin 
MODE NORMAL!
AUX HIGH!
Complete!
MODE SLEEP CONFIG!
AUX HIGH!
Complete!
3
Available buffer: 3 structure size: 11
----------------------------------------
HEAD : E1 40 C4
 
AddH : FE
AddL : BD
 
Chan : 205 -> 615MHz
 
SpeedParityBit     : 10 -> 8E1
SpeedUARTDatte     : 10 -> 4800bps
SpeedAirDataRate   : 1 -> 2.4kbps
 
OptionSubPacketSett: 0 -> 200bytes (default)
OptionTranPower    : 1 -> 17dBm
OptionRSSIAmbientNo: 0 -> Disabled (default)
 
TransModeWORPeriod : 101 -> 3000ms
TransModeEnableLBT : 1 -> Enabled
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeFixedTrans: 1 -> Fixed transmission (first three bytes can be used as high/low address and channel)
----------------------------------------
MODE NORMAL!
AUX HIGH!
Complete!
Data size not match!
7
----------------------------------------
HEAD : E1 40 C4
 
AddH : FE
AddL : BD
 
Chan : 205 -> 615MHz
 
SpeedParityBit     : 10 -> 8E1
SpeedUARTDatte     : 10 -> 4800bps
SpeedAirDataRate   : 1 -> 2.4kbps
 
OptionSubPacketSett: 0 -> 200bytes (default)
OptionTranPower    : 1 -> 17dBm
OptionRSSIAmbientNo: 0 -> Disabled (default)
 
TransModeWORPeriod : 101 -> 3000ms
TransModeEnableLBT : 1 -> Enabled
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeFixedTrans: 1 -> Fixed transmission (first three bytes can be used as high/low address and channel)
----------------------------------------
MODE SLEEP CONFIG!
AUX HIGH!
Complete!
3
Available buffer: 3 structure size: 6
MODE NORMAL!
AUX HIGH!
Complete!
Data size not match!
7
----------------------------------------
HEAD: C1 8 3
Model no.: 65
Version  : 64
Features : 20
----------------------------------------
```

blacxcode, 2023-05-17:
Now it's fine, just a problem of insufficient current in the device module.

xreef, 2023-05-17:
As said via mail can be power supply problem.
Bye Renzo

closed 2023-05-17 by xreef

Tatbiq-Tech, 2024-07-30:
@blacxcode I know its been a while, but have you succeeded to enable receiver and transmitter modes via esp-idf with esp and lora ? 

Thanks