xreef/EByte_LoRa_E220_Series_Library #15: how to set the module to deep sleep

homonto, 2023-06-16:
First I would like to say thank you for your great job.
Second: E220 module is capable of going to deep sleep.
But I am missing such method in your library.
Could you please advise how to send E220 to deep sleep and wake it then later?
I would assume by providing HIGH/LOW to M0 and M1 right, but do I need to do it on my own (digitalWrite(M0, HIGH/LOW) or it is there already?

thank you

homonto, 2023-06-16:
so I added to my code:

```
pinMode(AUX, INPUT);
pinMode(M0, OUTPUT);
pinMode(M1, OUTPUT);
```

and then:
```
void sleep()
{
  digitalWrite(M0, 1);
  digitalWrite(M1, 1);
}

void wakeup()
{
    digitalWrite(M0, 0);
    digitalWrite(M1, 0);
    Serial.print("Checking AUX status...");
    while (digitalRead(AUX) == 0) {}
    Serial.println("OK");
}
```

I think that does the job - at least I can see it does ;-)

xreef, 2023-06-16:
Hi homonto,
for example It's important that you specify the microcontroller; I write some articles about sleep and WOR.

For E220.
[Here ](https://mischianti.org/2022/05/06/ebyte-lora-e220-device-for-arduino-esp32-or-esp8266-manage-wake-on-radio-and-sends-structured-data-5/)you can check the management of sleep of the LoRa module, but without the control of microcontroller.

For the E32 that uses the equivalent library, I developed (something changed but not the management of the microcontroller).
[Here ](https://mischianti.org/2021/04/14/ebyte-lora-e32-device-for-arduino-esp32-or-esp8266-wor-wake-on-radio-and-new-esp32-shield-8/)you can check how to manage and persist the pin values of M0 and M1 with an esp32, [here ](https://mischianti.org/2020/01/17/lora-e32-device-for-arduino-esp32-or-esp8266-wor-wake-on-radio-microcontroller-and-new-wemos-d1-mini-shield-part-7/)with an esp8266 and [here ](https://mischianti.org/2019/12/28/lora-e32-device-for-arduino-esp32-or-esp8266-wor-wake-on-radio-the-microcontroller-also-and-new-arduino-shield-part-6/)with Arduino UNO

Tell me if you can find your needs. 

And if you write to the support forum, I think you can find more help.

Bye Renzo

homonto, 2023-06-16:
thank you
Since I will use LoRa for my remote sensors, I actually don't need WOR - I just want to:
1- wake up ESP32 from deep sleep
2- provide M0 and M1 LOW to wake E220 completely
3- check sensors
4- send data over LoRa
5- provide M0 and M1 HIGH to put to sleep E220 completely
6- deep sleep for ESP32 until next cycle

this way I would save a lot of power - E220 power consumption I just measured:
TX mode:
[image]

RX mode:
[image]

Sleep mode:
[image]

One thing I am surprised is the time it takes to send the message: from the graph we can see around 340ms while millis() show ... 20ms - I am not sure where the catch is - it is kind of 17 times more ;-)

homonto, 2023-06-16:
what I am trying to say is: if you want to go to sleep (E220) without checking AUX, you might end up thinking message was sent, while in fact it was not sent and "ResponseStatus rs" does NOT help with that at all.
That is my finding

xreef, 2023-06-17:
Hi,
Yes it's true, if you don't set Aux, I use an estimated time that cover all case (I try to use less time but I discover a lot of issue), and it's surely not so efficient, to get better efficiency you must connect AUX.
Bye Renzo

closed 2023-06-17 by homonto