xreef/EByte_LoRa_E220_Series_Library #16: is AUX necessary?

garudaonekh, 2023-06-21:
HI,
As per your description, AUX is not required.
But without AUX, available() function return 0. availabe() is fine when I use AUX pin

I use ESP32 custom PINs.

xreef, 2023-06-21:
Hi,
AUX isn't necessary, but isn't efficient because AUX advise that the operation is successful executed if isn't connected I wait a long time that cover all possible operations, so It waste a lot of time.
Bye Renzo

garudaonekh, 2023-06-22:
Ok, I set 1 second timeout to wait for reply.  My receiver immediately response back. Can we adjust your wait time? I do not have any pin left for AUX

xreef, 2023-06-22:
Hi,
the library already have inside a tested wait that cover all situation.
You don't add any other wait.
Bye Renzo

closed 2023-06-22 by xreef