xreef/EByte_LoRa_E220_Series_Library #18: Transmission not working with STM32 L432KC

paddygoat, 2023-09-26:
`#include "Arduino.h"
#include "LoRa_E220.h"// With FIXED SENDER configuration
#define DESTINATION_ADDL 3

// ---------------- STM32 --------------------
HardwareSerial Serial2(USART2);   // PA3  (RX)  PA2  (TX)
LoRa_E220 e220ttl(&Serial2, PA0, PB0, PA10); //  RX AUX M0 M1
// -------------------------------------------------

void printParameters(struct Configuration configuration);
void printModuleInformation(struct ModuleInformation moduleInformation);

void setup() {
	Serial.begin(9600);
  Serial.setTimeout(1000);
	while(!Serial){};
	delay(500);

	Serial.println();

	// Startup all pins and UART
	e220ttl.begin();
  Serial.println("e220ttl has begun");

	ResponseStructContainer c;
	c = e220ttl.getConfiguration();
	// It's important get configuration pointer before all other operation
	Configuration configuration = *(Configuration*) c.data;
	Serial.println(c.status.getResponseDescription());
	Serial.println(c.status.code);

	printParameters(configuration);

	ResponseStructContainer cMi;
	cMi = e220ttl.getModuleInformation();
	// It's important get information pointer before all other operation
	ModuleInformation mi = *(ModuleInformation*)cMi.data;

	Serial.println(cMi.status.getResponseDescription());
	Serial.println(cMi.status.code);

	printModuleInformation(mi);

	c.close();
	cMi.close();

  Serial.println("Hi, I'm going to send message!");

  // Send message
  ResponseStatus rs = e220ttl.sendFixedMessage(0, DESTINATION_ADDL, 23, "Hello, world?");
  // Check If there is some problem of succesfully send
  Serial.println(rs.getResponseDescription());
}

void loop() {

}
void printParameters(struct Configuration configuration) {
	Serial.println("----------------------------------------");

	Serial.print(F("HEAD : "));  Serial.print(configuration.COMMAND, HEX);Serial.print(" ");Serial.print(configuration.STARTING_ADDRESS, HEX);Serial.print(" ");Serial.println(configuration.LENGHT, HEX);
	Serial.println(F(" "));
	Serial.print(F("AddH : "));  Serial.println(configuration.ADDH, HEX);
	Serial.print(F("AddL : "));  Serial.println(configuration.ADDL, HEX);
	Serial.println(F(" "));
	Serial.print(F("Chan : "));  Serial.print(configuration.CHAN, DEC); Serial.print(" -> "); Serial.println(configuration.getChannelDescription());
	Serial.println(F(" "));
	Serial.print(F("SpeedParityBit     : "));  Serial.print(configuration.SPED.uartParity, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getUARTParityDescription());
	Serial.print(F("SpeedUARTDatte     : "));  Serial.print(configuration.SPED.uartBaudRate, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getUARTBaudRateDescription());
	Serial.print(F("SpeedAirDataRate   : "));  Serial.print(configuration.SPED.airDataRate, BIN);Serial.print(" -> "); Serial.println(configuration.SPED.getAirDataRateDescription());
	Serial.println(F(" "));
	Serial.print(F("OptionSubPacketSett: "));  Serial.print(configuration.OPTION.subPacketSetting, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getSubPacketSetting());
	Serial.print(F("OptionTranPower    : "));  Serial.print(configuration.OPTION.transmissionPower, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getTransmissionPowerDescription());
	Serial.print(F("OptionRSSIAmbientNo: "));  Serial.print(configuration.OPTION.RSSIAmbientNoise, BIN);Serial.print(" -> "); Serial.println(configuration.OPTION.getRSSIAmbientNoiseEnable());
	Serial.println(F(" "));
	Serial.print(F("TransModeWORPeriod : "));  Serial.print(configuration.TRANSMISSION_MODE.WORPeriod, BIN);Serial.print(" -> "); Serial.println(configuration.TRANSMISSION_MODE.getWORPeriodByParamsDescription());
	Serial.print(F("TransModeEnableLBT : "));  Serial.print(configuration.TRANSMISSION_MODE.enableLBT, BIN);Serial.print(" -> "); Serial.println(configuration.TRANSMISSION_MODE.getLBTEnableByteDescription());
	Serial.print(F("TransModeEnableRSSI: "));  Serial.print(configuration.TRANSMISSION_MODE.enableRSSI, BIN);Serial.print(" -> "); Serial.println(configuration.TRANSMISSION_MODE.getRSSIEnableByteDescription());
	Serial.print(F("TransModeFixedTrans: "));  Serial.print(configuration.TRANSMISSION_MODE.fixedTransmission, BIN);Serial.print(" -> "); Serial.println(configuration.TRANSMISSION_MODE.getFixedTransmissionDescription());

	Serial.println("----------------------------------------");
}
void printModuleInformation(struct ModuleInformation moduleInformation) {
	Serial.println("----------------------------------------");
	Serial.print(F("HEAD: "));  Serial.print(moduleInformation.COMMAND, HEX);Serial.print(" ");Serial.print(moduleInformation.STARTING_ADDRESS, HEX);Serial.print(" ");Serial.println(moduleInformation.LENGHT, DEC);

	Serial.print(F("Model no.: "));  Serial.println(moduleInformation.model, HEX);
	Serial.print(F("Version  : "));  Serial.println(moduleInformation.version, HEX);
	Serial.print(F("Features : "));  Serial.println(moduleInformation.features, HEX);
	Serial.println("----------------------------------------");
}`

This above code gives error:
Arduino: 1.8.19 (Linux), Board: "Nucleo-32, Nucleo L432KC, Mass Storage, Enabled (generic 'Serial'), None, Low/Full Speed, Smallest (-Os default), None, Newlib Nano (default)"

/home/flatus/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/12.2.1-1.2/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_cache_824242/core/core_2c23ace741d1764fd7c6b8218f9f5d4c.a(HardwareSerial.cpp.o):(.bss.Serial2+0x0): multiple definition of `Serial2'; sketch/test_LORA_E220_05.ino.cpp.o:(.bss.Serial2+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Nucleo-32.

I then removed 'HardwareSerial Serial2(USART2);   // PA3  (RX)  PA2  (TX)' and the code then compiled, but get no transmit and this is output in terminal:
`e220ttl has begun?
No response from device! (Check wiring)
12
----------------------------------------
HEAD : 0 2C 0
 
AddH : D1
AddL : 9
 
Chan : 189 -> 599MHz
 
SpeedParityBit     : 0 -> 8N1 (Default)
SpeedUARTDatte     : 1 -> 2400bps
SpeedAirDataRate   : 0 -> 2.4kbps
 
OptionSubPacketSett: 0 -> 200bytes (default)
OptionTranPower    : 0 -> 22dBm (Default)
OptionRSSIAmbientNo: 0 -> Disabled (default)
 
TransModeWORPeriod : 0 -> 500ms
TransModeEnableLBT : 1 -> Enabled
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeFixedTrans: 0 -> Transparent transmission (default)
----------------------------------------
⸮No response from device! (Check wiring)
12
----------------------------------------
HEAD: 0 0 0
Model no.: 0
Version  : 73
Features : 70
----------------------------------------
Hi, I'm going to send message!
Hello, World?Success
`
Please advise !!!! .... Thank you.

xreef, 2023-09-27:
Hi,
It seems, that there is a wiring problem.
The device not move the AUX pin.
Bye Renzo

paddygoat, 2023-09-28:
Thanks for reply Renzo.
I checked my wiring, especially the AUX on PA0 and wrote the following lines to check it:
```
void setup() {
  pinMode(PA0,OUTPUT);
  digitalWrite(PA0,LOW);
  delay(500);
```
... which produced a 500ms pulse on scope as expected, but nothing else.

I then added this code:
```
  // Startup all pins and UART
  e220ttl.begin();
  digitalWrite(PA0,LOW);
  delay(500);
  digitalWrite(PA0,HIGH);
  delay(500);
```
.... which produced no pulse at all.
Removing pinMode(PA0,OUTPUT); made no difference.

Not sure what is going on !!!!

Tech500, 2026-11-01:
Power off; use digital multimeter to check ground continuity between exposed metal ( USB connector,  exposed metal to STM32 ground pin).   When ground is confirmed to be good; restore power; measure Vcc rail to verify proper, Vcc voltage .  

Try library example code for your STM32 L432KC devices.  Will isolate if it is code or wiring.  

If breadboarding; both Vcc and Ground power rails may have  no continous continuity end to end and need a jumper where there is an open (common with breadboards).