xreef/EByte_LoRa_E220_Series_Library #19: Does it support E220P?

Catta1997, 2023-09-27:
Hi! I have an issue with E220P-400T22S , I'm not able to send/receive messages . This is the code:
```
#include "Arduino.h"
#include "LoRa_E220.h"

LoRa_E220 e220ttl(DD3, DD2, DD4, DD6, DD5);

void setup() {
    Serial.begin(115200);
    delay(500);
    e220ttl.begin();
    Serial.println("Setup Done!");
}

void leggo() {
    if (e220ttl.available() > 1) {
        Serial.println("Miracolo, ricevuto qualcosa!");
        ResponseContainer rc = e220ttl.receiveMessage();
        if (rc.status.code != 1) {
            Serial.println(rc.status.getResponseDescription());
        } else {
            Serial.println(rc.status.getResponseDescription());
            Serial.println(rc.data);
        }
    }
}

void mando() {
    ResponseStatus rs = e220ttl.sendMessage("Messaggio di prova");
    Serial.println(rs.getResponseDescription());
    delay(2500);
}

void loop() {
    // leggo(); -> read
    // mando(); -> write
}
```

Setup log:
```
RX MIC ---> 3
TX MIC ---> 2
AUX ---> 4
M0 ---> 6
M1 ---> 5
Init AUX pin!
Init M0 pin!
Init M1 pin!
Begin ex
Begin Software Serial
Begin 
MODE NORMAL!
AUX HIGH!
Complete!
Setup Done!
```

Sender log: 
```
Send message: Messaggio di prova size: 18
 memcpy 
AUX HIGH!
Complete!
Clear buffer...ok!
Success
```
Rec Log:
nothing more than setup log

I'm doing something wrong?

xreef, 2023-09-27:
Hi,
I don't know witch mcu you have but normally people fix this by check the power supply of e220 and add an antenna.
Bye Renzo

Catta1997, 2023-09-30:
ok thanks

Catta1997, 2023-09-30:
I printed all the config using your example and I discovered something strange, 
`Chan : 174 -> 584MHz`

Why is it using 584? 584 is neither an option of `OPERATING_FREQUENCY` and in the file `StatesNaming.h` OPERATING_FREQUENCY is defined 410 (default) . Any idea? I'm not using any `setConfiguration()`