xreef/EByte_LoRa_E220_Series_Library #22: Inconsistent data in reception with Uno R4 Minima

WRoss7, 2024-02-23:
I am experiencing problems in receiving with the Arduino R4 Minima board, while the same sketch loaded on Arduino Uno R3 works correctly; specifically, the data coming in is not consistent. In transmission everything works correctly.
I have done many tests including;
- reprogrammed the LoRa module
- tried 2 other LoRa modules that worked perfectly
- tried various usb cables
- tested the sketch on the library site
- ...
but without success

Json sent by an esp32:
Payload: {
    "dispositivo": "home assistant  ",
    "presenzaPioggia": "255",
    "bloccoIrrigazione": "255",
    "temperaturaMassima": "28.4",
    "statoDispositivo": "255",
    "erroreTx": "255",
    "loraRssi": "255",
    "wifiRssi": "255"
}

Data received from Arduino R4 Minima:
     - Dispositivo: home assistant  
     - Presenza pioggia: 50
     - Blocco irrigazione: 56
     - Temperatura massima.: .4
     - Stato dispositivo: 40
     - Errore Tx: 73
     - LoRa RSSI: 0
     - WiFi RSSI: 251

Data received from Arduino Uno R3 (R4 Minima replacement and same connections):
     - Dispositivo: home assistant  
     - Presenza pioggia: 255
     - Blocco irrigazione: 255
     - Temperatura massima.: 28.4
     - Stato dispositivo: 1
     - Errore Tx: 255
     - LoRa RSSI: 240
     - WiFiRSSI: 171

The first data item (Dispositivo) is always correct.
The last 4 data are taken from the transmission.
The other three data (Presenza pioggia, Blocco irrigazione, e Temperatura massima) are critical to me, but they are not correct.

Structure in the receiver:
  char MessaggioDispositivo[17];
  byte MessaggioPresenzaPioggia;
  byte MessaggioBloccoIrrigazione;
  char MessaggioTemperaturaMax[6];
  byte MessaggioStatoDispositivo;
  byte MessaggioErroreTx;
  byte MessaggioLoRaRSSI;
  byte MessaggioWiFiRSSI

In the transmitter I have all the pins of the LoRa module connected, while in the receiver I connected only Tx and Rx (as in the diagram on the library website). However, I also tried connecting the remaining pins, but nothing changes..

Am I doing something wrong?
Does anyone know how I can solve the problem?

xreef, 2024-02-24:
Hi Wross,
It's strange, it seems that the size of the variable was different from the devices, but you use primitive type that can't generate that problem.
Try to send simple message without structure and give us a feedback.
Bye Renzo

WRoss7, 2024-02-24:
Ciao Renzo,
thank you for the quick response.

The code for the transmitter:
  ResponseStatus TrasmissioneMessaggio = e220ttl.sendFixedMessage(Indirizzo_ADDH, Indirizzo_ADDL, Channel, "Sending this message Hello World");

And for the receiver:
  ResponseContainer RicezioneMessaggio = e220ttl.receiveMessage();
  Serial.println(RicezioneMessaggio.status.getResponseDescription());
  Serial.println(RicezioneMessaggio.data);

And this is the result:
  Success
  Sending this message Hello World

It works as well as when I send the device name with the first internal variable in the structure.

Other tests...
if I send data with value 0-:-127, and remove the temperature one, I get correct data in the struct.

I hope it can be helpful to understand where the problem lies.

WRoss7, 2024-03-06:
Ciao Renzo,

test with the "complete sender and receiver sketch" found at "https://mischianti.org/it/ebyte-lora-e220-llcc68-per-arduino-esp32-o-esp8266-power-saving-ed-invio-di-dati-strutturati-5/"

And below are the results:

Transmission from Arduino Uno R4 Minima to Esp32
TEMP
19.20
Kitchen
RSSI: 242
All data arrive correctly

Transmission from Esp32 to Arduino Uno
TEMP
19.20
Kitchen
RSSI: 237
All data arrive correctly

**********************************************************
Transmission from Esp32 to Arduino Uno R4 Minima
TEMP
0.00
Kitchen
RSSI: 124
Both temperature and RSSI signal data are incorrect
**********************************************************

Could this be due to an incompatibility between the library and Arduino Uno R4?

Unfortunately, my sketch does not fit in the Arduino Uno by size.
Let me know if you need any further testing

Walter

WRoss7, 2026-02-11:
The bug in Arduino Minima R4 has been fixed and everything is now working as it should.

closed 2026-02-11 by WRoss7