xreef/EByte_LoRa_E220_Series_Library #23: How to improve recieve and send times?

CcKefa, 2024-03-07:
Hello, i'm trying to build a system with 3 devices that comunicate between them in a cascade order. 1 sends info to 2 and 2 to 3. The problem that i'm having is when i send multiple packages of info from 1 to 2, 2 isn't capable of receiving and sending fast enough, resulting in half of the data missing.
The way i have it set up is the following:
`  if (e220ttl.available() > 1)
  {
    Serial.println("Mensaje recibido por LoRa!");
    // read the String message
    ResponseContainer rc = e220ttl.receiveMessage();
    if (rc.status.code != 1)
    {
      Serial.println(rc.status.getResponseDescription());
    }
    else
    {
      #ifdef DEBUG
        Serial.println(rc.data);      
        Serial.println("Enviando por LoRa");
      #endif
        String json = rc.data;
        ResponseStatus rs = e220ttl.sendFixedMessage(DESTINATION_ADDH, DESTINATION_ADDL, channel, json);
  }`

So as you can see, it basically checks constantly if there is data available from the LoRa device and sends it to the next device.
I think the problem that i'm having is because while the device is sending info, it's recieving a package at the same time, resulting in lost data. The device prior to this one, sends about 60 bytes every second. I already adjusted the Air Data Rate and sub packet size accordingly. Range is not an issue since the devices are relatively close.
I've ran out of ideas on what to try next.

closed 2024-03-11 by xreef

CcKefa, 2024-03-11:
Hello, i do have the AUX pin connected. What do you mean by FEC? I already adjusted air data rate and sub packet size. I will increase the baud rate and try.
Kind regards

xreef, 2024-03-12:
Sorry for e220 LBT
https://mischianti.org/ebyte-lora-e220-llcc68-device-for-arduino-esp32-or-esp8266-library-2/#Monitor_data_before_transmission

CcKefa, 2024-03-13:
Hello, i already read all the post. I was wondering if there is something else i could be doing to improve the recieve-send times. Is there something in the code that could be improven?

Moskus, 2024-04-10:
We just ran into this problem. In the end we found the problem in LoRa_E220.cpp on line 826:

`String tmpData = this->serialDef.stream->readString();`
... will always take 102 ms (in our setup). The culprit is "timedRead()" in Stream.cpp from espriff, which always times out it seems.

We did a quick check and started to end our messages with "#", and updated line 826 to:
`String tmpData = this->serialDef.stream->readStringUntil('#');`

This reduced the time receivedMessageComplete took from 102 ms to 2 ms. 😊

However, this is not an ideal solution as there might be more messages waiting in the buffer.

CcKefa, 2024-04-12:
I managed to lower the recieve and send time from more than a second to 200ms by using 2 devices and running recieve and send functions in the background. If you are looking for speed this seems the best way to do it but not cost effective.