xreef/EByte_LoRa_E220_Series_Library #24: E220 modules stop working?

Moskus, 2024-03-13:
We have 10 modules, and when they work, they are working great. We have 2 modules sending data every second to a receiver.

However, all of the sudden and out of the blue, the modules just stop working. When this happens it doesn't matter what we do, we can't make the module work again. But if we wait a few hours or until the next day, the module most likely will work again.

We connected the AUX port to a a working module to an oscilloscope. As we are sending data using a different module, we can easily see that the AUX port is doing what it should do:
[image]

If we connect a non-working module, we only get a flat line:
[image]

Are we the only ones with this problem?
Is there anything we can do about it?

xreef, 2024-03-19:
Mmm.. I think It's broken; try to connect as Serial and use the EByte program to reset.
But I think it isn't useful.
Bye Renzo

Moskus, 2024-03-21:
Setting the config with the EByte program solved the problem. As long as we don't try to update the config using this library, the modules work just fine.

closed 2025-01-10 by xreef