xreef/EByte_LoRa_E220_Series_Library #25: Support for other brands of LLCC68

LimesKey, 2024-05-30:
I bought a few of the DL-LLCC68-S 915MHz LORA modules and I'm planning to use it with an ESP32. I see that this library supports the EByte LLCC68 but not the DreamLNK one. Is there cross-compatibility with this library? I have a photo of the `DL-LLCC68-S-915` with a custom PCB I made for it to be able to use standard 2.54mm Dupont wires.

[image]

[image]

I didn't buy the exact one from [LCSC ](https://www.lcsc.com/product-detail/LoRa-Modules_DreamLNK-DL-LLCC68-S-915_C2894791.html) but I bought a similar or clone one from AliExpress. Hope you can help as these LORA modules are relatively cheap at only 3$ and it'd be amazing to use them.

LimesKey, 2024-05-31:
With some more research, I see that the DreamLNK version of the LLCC68 has SPI instead of UART like this library uses, would it be possible to buy something like [this  ](https://www.sparkfun.com/products/retired/9981) to convert SPI to UART?

xreef, 2024-06-03:
Hi @LimesKey,
no, the UART version of E220 has an MCU that manages a lot of aspects of the device, like signal check, encryption, and other features.
Bye Renzo

LimesKey, 2024-06-03:
Would it be possible and easy enough for someone like me to fork your code and adapt it to work for this SPI LORA module?