xreef/EByte_LoRa_E220_Series_Library #27: Stop working with ESP32 board manager 3.x.x (based on ESP-IDF v.5)

lovefool, 2024-06-26:
My program has been working and suddenly stopped. The cause is updating ESP32 board manager from 2.017 to 3.0.0. 
Since new ESP32 board manager is based on ESP-IDF v5 and I know a lot have been changed.
Currently I changed ESP32 board manager back to 2.0.17, and my program started working again.

gabrielgrdp, 2024-06-28:
Hey mate, does it still work on this version of the board?

Here the transmission stopped working, I went back to the version I was using (2.0.11) and today it stopped working again, even though I tested your version I was unable to communicate.

lovefool, 2024-06-29:
I have changed ESP32 board manager back 2.0.17, then it started working again. I tested on two environment Win11 22H2 and MacOS, both now working again.

xreef, 2024-07-28:
I think it's related to #28, so please check it.
Thanks Renzo

xreef, 2024-07-28:
I think the problem can be this

[image]

from core 2 to core 3 TX2 and RX2 pins are changed 

try to force the pin number
LoRa_E32 e32ttl100(16, 17, &Serial2, 15, 21, 19, UART_BPS_RATE_9600); //  RX AUX M0 M1

Bye Renzo

lovefool, 2024-08-03:
Thank you for your anlysis. I will check. 
BTW, U2Txd/U3Txd pins are 17/16, I don't know why core3 assign different pins. pin swap?