xreef/EByte_LoRa_E220_Series_Library #29: Arduino Uno R3 - Timeout

Nikolay961996, 2024-10-19:
Hello. You did a great job! But i have problem with Arduino Uno R3 with E220-900T22D. I wire like you pic, and run you code [01_setConfiguration](https://github.com/xreef/EByte_LoRa_E220_Series_Library/tree/master/examples/01_setConfiguration) with adding on top
```
#define FREQUENCY_868
#define LoRa_E220_DEBUG
```
and module info printing
```
e220ttl.begin();
ResponseStructContainer c;
c = e220ttl.getModuleInformation();
ModuleInformation moduleInformation = *(ModuleInformation*) c.data;
printModuleInformation(moduleInformation);
```

I get next message:
```
----------------------------------------
HEAD: 0 0 2
Model no.: 0
Version  : 50
Features : A1
----------------------------------------
Timeout!!
9
----------------------------------------
----------------------------------------
HEAD: 0 0 2
Model no.: 0
Version  : 50
Features : A1
----------------------------------------
Timeout!!
9
----------------------------------------
HEAD : 0 0 2
 
AddH : 0
AddL : 50
 
Chan : 0 -> 850MHz
 
SpeedParityBit     : 0 -> 8N1 (Default)
SpeedUARTDatte     : 101 -> 38400bps
SpeedAirDataRate   : 1 -> 2.4kbps
 
OptionSubPacketSett: 0 -> 200bytes (default)
OptionTranPower    : 11 -> 10dBm
OptionRSSIAmbientNo: 0 -> Disabled (default)
 
TransModeWORPeriod : 0 -> 500ms
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeFixedTrans: 0 -> Transparent transmission (default)
----------------------------------------
Timeout!!
9
Timeout!!
9
----------------------------------------
HEAD : 0 0 2
 
AddH : 0
AddL : 50
 
Chan : 0 -> 850MHz
 
SpeedParityBit     : 0 -> 8N1 (Default)
SpeedUARTDatte     : 101 -> 38400bps
SpeedAirDataRate   : 1 -> 2.4kbps
 
OptionSubPacketSett: 0 -> 200bytes (default)
OptionTranPower    : 11 -> 10dBm
OptionRSSIAmbientNo: 0 -> Disabled (default)
 
TransModeWORPeriod : 0 -> 500ms
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeFixedTrans: 0 -> Transparent transmission (default)
----------------------------------------
```

Nikolay961996, 2024-10-19:
I measure now on Lora pins:
M0, M1, TXD, AUX - 3.3V
RXD - 1V
VCC+GND - 5V

Nikolay961996, 2024-10-19:
I found problem. Arduino wiring pin picture in ReadMe not some pins in scratch

xreef, 2025-01-10:
Thanks for your feedback!
bye Renzo

closed 2025-01-10 by xreef