xreef/EByte_LoRa_E220_Series_Library #30: In 05_sendFixedTransmissionStructure missed rsc.close()?

FedeJack06, 2024-11-02:
Hi, in the example 05_sendFixedTransmissionStructure, the ResponseStructContainer rsc must be closed right?

xreef, 2024-11-04:
Hi Federico,
yes, you are right. 
Bye Renzo

closed 2025-01-10 by xreef

ilyaborisov21, 2025-02-01:
Hello, 
Please.
Can you clarify how you fixed it? Incorrect data comes when using the structure, the compiler does not write about errors.

ilyaborisov21, 2025-02-01:
Something goes wrong!!
MI
Something goes wrong!!
throo�?
Something goes wrong!!

Something goes wrong!!
TEMP
19.20
Bathroo

Here MI are all that remains of HUMI. First part is loss

FedeJack06, 2025-02-10:
```
struct radioVal {};
radioVal radioRx;
bool checkReceiveLLCC68() {
	if (e220ttl.available()>1) {
		ResponseStructContainer rsc = e220ttl.receiveMessage(sizeof(radioRx));
		Serial.println(F("riceved"));
		if (rsc.status.code!=1){
			Serial.println(rsc.status.getResponseDescription());
			return 0;
		}else{
			Serial.println(rsc.status.getResponseDescription());
			radioRx = *(radioVal*) rsc.data;
			rsc.close();
			return 1;   
		}
	} else
		return 0;
}